wsh32/airhockey #28: Puck tracker doesn't always capture puck

wsh32, 2021-11-02:
Depending on lighting, the puck isnt always tracked.

Case where it works:
[image]

Case where it doesn't work:
[image]

We may need to either figure out a way to have consistent lighting or recalibrate the color tracker. The HSV values are currently found in `src/airhockey_vision/config/detect_green.yaml`.

wsh32, 2021-11-02:
Also sometimes it doesnt capture the center correctly
[image]

wsh32, 2021-11-02:
[images.zip](https://github.com/wsh32/airhockey/files/7464582/images.zip)

Heres a bunch of test images

iserrato, 2021-11-02:
Lighting would help with this issue significantly

wsh32, 2021-11-16:
switched to a matte blue vinyl cut puck: 
[image]

Looks like this issue is mostly resolved. Note that there are a couple of odd cases where the system picks up the blue "Air" text instead of the puck, but this was fixed by coloring in that light blue with a black sharpie

PR #77 (2021-11-16): Make puck tracker track the light blue vinyl pucks
resolves #28
closed 2021-11-16 by wsh32